GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader #1: [ISSUE] ST-link claims Read Out Protection is enabled

shuffleznl, 2021-03-04:
### Bug Description
First of all thanks for your great work, doing what Biqu/BTT failed to!
Makes me very humbled to see the work individuals put in to support others.
I'm not sure if to post here, as it's not a bug with any of your doing at all :)

Issue happened after abruptly disconnecting board (during print, so FW/SW was running well).
Rebooting the board didn't start nor did it read from SD (tried several FW's).

Upon connection of ST-link (swapped SWCLK & SWDIO to be sure) it reads Device, DeviceID, RevisionID, but Flash Size is reported 'Unknown'. Also a warning is thrown; '_Can not read memory! Disable Read Out Protection and retry._'

Cannot reset Option Bytes Read Out Protection from 1 to 0.
Full Chip Erase is also 'protected' because of '_Read Out protection is activated_' warning.

#### Additional Information
[image]
[image]

GadgetAngel, 2021-03-04:
@shuffleznl Are you using the ORIGINAL or OFFICAL ST-LINK-V2 device or a clone device? Which BIGTREETECH are you trying to talk to?

GadgetAngel, 2021-03-04:
@shuffleznl Try to recover the chip by using the system bootloader:

Force boot0 pin to be 1 and force boot1 pin to be 0 at power up, start bootloader then connect USB and program the chip with DFU programmer. You can download the DFU programmer https://www.st.com/en/development-tools/stsw-stm32080.html#

ngandalf, 2022-01-15:
i am in same situation with my GTR bigtreetech 1.0 :-( 
@shuffleznl have you fix this situation ?

ngandalf, 2022-01-15:
on linux i had this error :

nicolas@nicolas-PC:~$ st-info --probe
Found 1 stlink programmers
 serial: 530041001500004d32414d4e
openocd: "\x53\x00\x41\x00\x15\x00\x00\x4d\x32\x41\x4d\x4e"
  flash: 4201472 (pagesize: 16384)
   sram: 196608
 chipid: 0x0413
  descr: F4 device
nicolas@nicolas-PC:~$ 

nicolas@nicolas-PC:~$ st-flash --reset --format=ihex write GTR-V1.0-bootloader-only.hex 
st-flash 1.6.0
2022-01-16T00:14:17 INFO usb.c: -- exit_dfu_mode
2022-01-16T00:14:17 INFO common.c: Loading device parameters....
2022-01-16T00:14:17 INFO common.c: Device connected is: F4 device, id 0x10076413
2022-01-16T00:14:17 INFO common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x401c00 bytes (4103 KiB) in pages of 16384 bytes
2022-01-16T00:14:17 INFO common.c: Attempting to write 21628 (0x547c) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08004000 erasedEraseFlash - Sector:0x1 Size:0x4000 
2022-01-16T00:14:17 INFO common.c: Finished erasing 2 pages of 16384 (0x4000) bytes
2022-01-16T00:14:17 INFO common.c: Starting Flash write for F2/F4/L4
2022-01-16T00:14:17 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 21628
2022-01-16T00:14:17 INFO common.c: Starting verification of write complete
2022-01-16T00:14:17 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1

nicolas@nicolas-PC:~$ st-flash --reset write GTR-V1.0-bootloader-only.bin  0x08000000
st-flash 1.6.0
2022-01-16T00:16:44 INFO common.c: Loading device parameters....
2022-01-16T00:16:44 INFO common.c: Device connected is: F4 device, id 0x10076413
2022-01-16T00:16:44 INFO common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x401c00 bytes (4103 KiB) in pages of 16384 bytes
2022-01-16T00:16:44 INFO common.c: Attempting to write 21628 (0x547c) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08004000 erasedEraseFlash - Sector:0x1 Size:0x4000 
2022-01-16T00:16:44 INFO common.c: Finished erasing 2 pages of 16384 (0x4000) bytes
2022-01-16T00:16:44 INFO common.c: Starting Flash write for F2/F4/L4
2022-01-16T00:16:44 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 21628
2022-01-16T00:16:44 INFO common.c: Starting verification of write complete
2022-01-16T00:16:44 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
nicolas@nicolas-PC:~$

can you help me please  ? i used a stlink V2 clone
is it possible that my GTR is dead?

GadgetAngel, 2022-01-16:
> on linux i had this error :
> 
> **st-info --probe** Found 1 stlink programmers serial: 530041001500004d32414d4e openocd: "\x53\x00\x41\x00\x15\x00\x00\x4d\x32\x41\x4d\x4e" flash: 4201472 (pagesize: 16384) sram: 196608 chipid: 0x0413 descr: F4 device
> 
> **st-flash --version** v1.6.0
> 
> **st-flash --reset --format=ihex write GTR-V1.0-bootloader-only.hex** st-flash 1.6.0 2022-01-16T00:14:17 INFO usb.c: -- exit_dfu_mode 2022-01-16T00:14:17 INFO common.c: Loading device parameters.... 2022-01-16T00:14:17 INFO common.c: Device connected is: F4 device, id 0x10076413 2022-01-16T00:14:17 INFO common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x401c00 bytes (4103 KiB) in pages of 16384 bytes 2022-01-16T00:14:17 INFO common.c: Attempting to write 21628 (0x547c) bytes to stm32 address: 134217728 (0x8000000) Flash page at addr: 0x08004000 erasedEraseFlash - Sector:0x1 Size:0x4000 2022-01-16T00:14:17 INFO common.c: Finished erasing 2 pages of 16384 (0x4000) bytes 2022-01-16T00:14:17 INFO common.c: Starting Flash write for F2/F4/L4 2022-01-16T00:14:17 INFO flash_loader.c: Successfully loaded flash loader in sram enabling 32-bit flash writes size: 21628 2022-01-16T00:14:17 INFO common.c: Starting verification of write complete 2022-01-16T00:14:17 ERROR common.c: Verification of flash failed at offset: 0 stlink_fwrite_flash() == -1
> 
> **st-flash --reset write GTR-V1.0-bootloader-only.bin 0x08000000** st-flash 1.6.0 2022-01-16T00:16:44 INFO common.c: Loading device parameters.... 2022-01-16T00:16:44 INFO common.c: Device connected is: F4 device, id 0x10076413 2022-01-16T00:16:44 INFO common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x401c00 bytes (4103 KiB) in pages of 16384 bytes 2022-01-16T00:16:44 INFO common.c: Attempting to write 21628 (0x547c) bytes to stm32 address: 134217728 (0x8000000) Flash page at addr: 0x08004000 erasedEraseFlash - Sector:0x1 Size:0x4000 2022-01-16T00:16:44 INFO common.c: Finished erasing 2 pages of 16384 (0x4000) bytes 2022-01-16T00:16:44 INFO common.c: Starting Flash write for F2/F4/L4 2022-01-16T00:16:44 INFO flash_loader.c: Successfully loaded flash loader in sram enabling 32-bit flash writes size: 21628 2022-01-16T00:16:44 INFO common.c: Starting verification of write complete 2022-01-16T00:16:44 ERROR common.c: Verification of flash failed at offset: 0 stlink_fwrite_flash() == -1
> 
> can you help me please ? i used a stlink V2 clone is it possible that my GTR is dead?

Instead of loading only the bootloader file, try flashing the "Bootloader+firmware" file.  This file contains the Micro-SD card bootloader and the original firmware that was delivered when you got the GTR board fresh from the factory.

Also check to see if the micro-SD card reader is even working.  I have seen other users say that the micro-SD card reader has bent pins. Use a flashlight and check to see what the micro-SD card reader's input slot looks like.  Are all the pins flat in the slot?

On Linux, you could use the STM32programming Utility. They make a version that will run under Mint Linux.  I use that one when I am on my Linux laptop.  The last time I used st-link from GitHub (if I am remembering correctly) I had trouble with flashing the MCU boards.  If you must use st-link from GitHub then check out their current issue and closed issues (https://github.com/stlink-org/stlink/issues). Someone else might have experienced this issue and might have solved it.  You might have to rollback to a different version of st-link (https://github.com/stlink-org/stlink/releases)

I know you are using a V2-clone on Linux but I highly recommend you buy and official ST-LINK-V2.  STMelectronics has some proprietary stuff in that device that the clones do not have. 

If you own BTT boards, it is worth the investment to buy a genuine ST-LINK-V2.  They are around $55 (https://www.amazon.com/Microcontrollers-circuit-debugger-programmer-emulator/dp/B00C7WDTLG).  You have to think of buying an ST-LINK-V2 original as if you need a specialized set of Wera Allen keys.  For example Wera TORX Allen Keys cost around $55 on amazon (https://www.amazon.com/Wera-05024480001-Multicolor-L-Key-TORX/dp/B07B7XWJZ6).

shuffleznl, 2022-02-04:
Apologies for abandoning the thread, I moved to SKR board (until it fried), on different (non-BTT) HW now.

Can confirm; Wera **and** original ST-Link are both worth it.