GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader #2: [QUESTION] Change bootloader

lutzmor, 2021-03-07:
Hey,

its not a bug more of a question :P, do you know if its possible to change the bootloader in a way that i can flash a klipper firmware to the board without the need to add a jumper cable on EXP1 and press reset? :)

Greetings

GadgetAngel, 2021-03-07:
> klipper firmware

Which BIGTREETECH board are you trying to use with Klipper? To Answer you question, it depends on the BIGTREETECH board

lutzmor, 2021-03-08:
Yeah indeed sry! Its a SKR 1.4 Turbo i flashed the bootloader and i'm able to flash it now but need to put the jumper wire in exp1 to port 0.28 and ground :).

Greetings

GadgetAngel, 2021-03-08:
> Yeah indeed sry! Its a SKR 1.4 Turbo i flashed the bootloader and i'm able to flash it now but need to put the jumper wire in exp1 to port 0.28 and ground :).
> 
> I currently let the wire in place so i'm able to flash the firmware on whenever i want to a bit hacky but it works i guess :).
> 
> Greetings

Here is the documentation I have found: https://www.klipper3d.org/Bootloaders.html.  

Flashing the bootloader is different from flashing the firmware.bin file.  The bootloader is the file located at 0x00000000 address in FLASH memory and examines the contents of the SD card to see if new firmware.bin file must be loaded into FLASH memory.  The bootloader.bin file is usually between 16KiB to 32KiB.  The firmware.bin file is the one you generate using Marlin. The size of the firmware.bin file depends on your printer and which features you turn on and off in Marlin.  
You only need to FLASH the bootloader.bin file only once while FLASHING the firmware.bin file will occur much much more often.  

You want to compile the bootloader for the SKR V1.4 Turbo and FLASH it once.  I have a bootloader code for the SKR V1.4 Turbo located here: https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader (see second purpose for the repository).  

The starting address for the bootloader is located in this document: https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/Bigtreetech%20boards%20Bootloaders_v2.0.pdf

The already have the compiled bootloader code located (https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/tree/main/bootloader_bin/compiled_bootloader_bin/SKR%20V1.4%20Turbo/Bootloader%20ONLY) will examine the contents of the SKR V1.4 Turbo's Micro-SD card to see if the contents of the firmware.bin file needs to be loaded into memory.   This way you will not need any jumper for flashing the firmware.bin file to the board anytime you recompile the Marlin code. All you will need to do is load the firmware.bin file to the Micro-SD card. I think you may even be able to upload the firmware.bin file via the USB cable directly for VScode.

The GUIDE for flashing the bootloader is located here: https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/tree/main/bootloader_bin/compiled_bootloader_bin/SKR%20V1.4%20Turbo. 

I hope this information is helpful.