GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader #3: [Question] - Possible to flash using arduino/RPI?

Section59, 2021-11-09:
As in the title, is it possible to flash the bootloaders using a raspberry or Arduino device or are the ST link devices necessary, just thought id ask beforehand as i have both an uno and multiple raspberries laying around and if i could not buy a device im likely to use once i'd rather do that if possible?

I have a skr 1.3 that is refusing to update firmware and has no connection via usb to my pc

GadgetAngel, 2021-11-09:
> As in the title, is it possible to flash the bootloaders using a raspberry or Arduino device or are the ST link devices necessary, just thought id ask beforehand as i have both an uno and multiple [...]

The SKR V1.3 and SKR V1.4 boards are LPC17xx based processor boards.  These boards must be placed in a **bootloader mode** before you can start to talk to the processor.  

To update the bootloader firmware on an SKR V1.3 board you will need to use a USB-to-TTL adapter.  If the raspberry pi can be used to play this role, then yes using a raspberry pi will work. 

You must use the "FlashMagic software to upload the bootloader firmware.  This is the only software that I have found that works with the SKR V1.3 board.  

Here are the guides: 
https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/bootloader_bin/compiled_bootloader_bin/SKR%20V1.3/Guide%20UART%20Flash%20SKR%20V1.3%20or%20SKR%20V1.4%20or%20SKR%20V1.4%20TURBO%20using%20Windows%2010.pdf

and

https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/bootloader_bin/compiled_bootloader_bin/SKR%20V1.3/Guide%20UART%20Flash%20SKR%20V1.3%20or%20SKR%20V1.4%20or%20SKR%20V1.4%20TURBO%20using%20USB%20to%20TTL%20adapter%20using%20Windows%2010.pdf

You can also recompile your bootloader firmware (I did this with my SKR V1.4 board and this is the reason that the repository was developed [use the source code in this repository (https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/tree/main/build)], if you have access to a LINUX machine: https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/tree/main/bootloader_bin/compiled_bootloader_bin

Good luck.

Section59, 2021-11-13:
Gave up and bought both an ST-Link v2 and a USB to TTL adapter however i cannot get the board to boot into bootloader mode at all?

Am i being dumb here? Hold the reset button, power on, jumper onto the boot pins (skr 1.3), release reset button, remove boot jumper, nothings happening?

GadgetAngel, 2021-11-13:
> Gave up and bought both an ST-Link v2 and a USB to TTL adapter however i cannot get the board to boot into bootloader mode at all?
> 
> Am i being dumb here? Hold the reset button, power on, jumper onto the boot pins (skr 1.3), release reset button, remove boot jumper, nothings happening?

You can not use the ST-LINK v2 for the SKR V1.3 board because the processor is an LPC1768.  You need the USB to TTL adapter (either BTT Writer or a generic one you buy from amazon).

I usually do the following:
1) wire up the BTT-Writer or the USB-to-TTL adapter to the SKR V1.3 board (see the manual for how to wire up the device to the board)
2) Power Up the SKR V1.3 board.
3) Startup the FlashMagic Software on my PC.  
4) In FlashMagic ensure "Device" says LPC1768.
5) In FlashMagic, select the right COM Port so that I am talking to the SKR V1.3 board.  When you plug in the USB-to-TTL adapter (or BTT Writer) your computer will make a sound indicating that the COM Port has been assigned to the USB-to-TTL adapter. Open Device Manager and check your "COM" ports.
6) In FlashMagic, select BAUD RATE "38400"
7) Now on the SKR V1.3 board, Press and hold the RESET button.
8)  Jumper boot pin (ISP_BOOT) for only a fraction of a second, by using a DUPONT Jumper cable. I connect the Female end of the DUPONT jumper cable to one of the two PINS on the BOOT Header on the SKR V1.3 board. I use the male end of the DUPONT jumper cable to touch the other PIN on the BOOT Header for a fraction of a second.  You only need to touch the other PIN for a fraction of a second!  DO NOT place a jumper across both PINS on the BOOT Header!  Once the boot PIN has been grounded for a fraction of a second (do not leave boot pin jumpered), I release the RESET button.
9) I then test to see if the board was put into Boot Mode by trying to access the flash memory of the SKR V1.3 board by going to FlashMagic's  "ISP" menu, select "READ". Look at "READ MEMORY" tab, select any READ block and see if you can access a memory block.
10) If FlashMagic does not update with the contents of the memory block, then you need to Ground the BOOT PIN for a little bit longer then you did the first time.  **Timing is very important here.**  The amount of time you ground the BOOT PIN  is one part of the timing.  The second part of the timing is the amount of time that passes between the Boot PIN being ground and the release of the RESET button. 
11) After you change the timing, try again to access a READ block via FlashMagic.  The only way to know if you got the timing correct is if FlashMagic can do a READ of a memory block.

I know this process can be frustrating but it does work.  Just be glad you do not own and SKR V1.4 or SKR V1.4Turbo, they have to access the BOOT PIN via a resistor R28, at least the SKR V1.3 board provides you with a BOOT header!

Section59, 2021-11-13:
Thanks for the replies, and sorry for being a hassle! Im not a programmer nor an electronics master so sorry for needing everything explaining like im five. Apologies.

However im pretty sure either my board is completely goosed or im missing something, however i've gone over the manual pdf provided multiple times and still cannot get it to work.

Im using a USB to TTL Adapter
I have a Com Port for said adapter
When i plug in the SKR, i get a usb notification stating usb device not recognized.

Tried multiple times to get the device into boot mode using the above method and still no connection, whenever i try to read a blocks the ttl device just shows an attempt to transmit data (red flashing led) on the TXD before failing to connect.

Im at a loss as to what to try next, tried slower tapping boot, tried faster, tried longer and quicker releases of the reset etc etc and it just refuses to do anything :/

[image]
[image]
[image]
)

GadgetAngel, 2021-11-17:
> Thanks for the replies, and sorry for being a hassle! Im not a programmer nor an electronics master so sorry for needing everything explaining like im five. Apologies.
>
> However im pretty sure either [...]

Sorry, I have not checked my email for the last 4 days. That is why you are getting this so late.

In my list of steps I DO NOT states you need to connect the SKR V1.3 board to your computer via USB. If the original problem with the SKR board is that it will not connect to your computer via USB then to me that indicates two possible problems. 

1) The SKR Board experienced an ESD while the user inserted a micro-SD card into the on board SD-card reader. These boards are very very sensitive to ESD and if you forgot to ground yourself before touching the board, then the board would receive an ESD which has scrambled the digital bits in the flash memory of the SKR board.  Your board was just unlucky enough that this scrambling occurred in the flash memory where the boot loader is stored and the boot loader was damaged.

2) the SKR Board has been damaged by EMI (ESD) and that it will never be recovered, you might have to buy a new board. 

When you perform the following steps, DO NOT connect the SKR V1.3 to you computer via USB cable.  Power the SKR V1.3 board via 12V DC power supply and ensure that the black/red "Power Selection" jumper on the board is set for PSU power (not USB power).

1. wire up the BTT-Writer or the USB-to-TTL adapter to the SKR V1.3 board (see the manual for how to wire up the device to the board)
2. Power Up the SKR V1.3 board via the PSU (do not power via the USB port)
3. Startup the FlashMagic Software on my PC.
4. In FlashMagic ensure "Device" says LPC1768.
5. In FlashMagic, select the right COM Port so that I am talking to the SKR V1.3 board. When you plug in the USB-to-TTL adapter (or BTT Writer) your computer will make a sound indicating that the COM Port has been assigned to the USB-to-TTL adapter. Open Device Manager and check your "COM" ports.
6. In FlashMagic, select BAUD RATE "38400"
7. Now on the SKR V1.3 board, Press and hold the RESET button.
8. Jumper boot pin (ISP_BOOT) for only a fraction of a second, by using a DUPONT Jumper cable. I connect the Female end of the DUPONT jumper cable to one of the two PINS on the BOOT Header on the SKR V1.3 board. I use the male end of the DUPONT jumper cable to touch the other PIN on the BOOT Header for a fraction of a second. You only need to touch the other PIN for a fraction of a second! DO NOT place a jumper across both PINS on the BOOT Header! Once the boot PIN has been grounded for a fraction of a second (do not leave boot pin jumpered), I release the RESET button.
9. I then test to see if the board was put into Boot Mode by trying to access the flash memory of the SKR V1.3 board by going to FlashMagic's "ISP" menu, select "READ". Look at "READ MEMORY" tab, select any READ block and see if you can access a memory block.
10. If FlashMagic does not update with the contents of the memory block, then you need to Ground the BOOT PIN for a little bit longer then you did the first time. Timing is very important here. The amount of time you ground the BOOT PIN is one part of the timing. The second part of the timing is the amount of time that passes between the Boot PIN being ground and the release of the RESET button.
11. After you change the timing, try again to access a READ block via FlashMagic. The only way to know if you got the timing correct is if FlashMagic can do a READ of a memory block.

The USB-to-TTL adapter should be recognized by your computer and assigned COM port. It sounded like your computer DID recognize the USB-to-TTL adapter.

If after multiple attempts of trying to get the SKR 1.3 board into boot mode [while powering the SKR board via the a power supply (not the USB port)] fails then I would buy another SKR V1.3 or SKR V1.4 board. The SKR V1.3/V1.4 boards are getting harder to find. You might want to consider using the SKR mini E3 series of boards.

I hope all this helps.