GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader #5: Klipper/Fluidd "Can not update MCU 'rpi' config as it is shutdown"

EmreDogann, 2022-01-19:
Hey so I'm using an SKR mini E3 Turbo with Klipper + Fluidd running on it and I managed to flash the pre-compiled bootloader only (no firmware) onto the board and everything seems to be working fine (i.e. Silent beeper, USB firmware uploads). However, now I have this issue where if I leave the Klipper service running on the RPi and power cycle the printer via the power switch on the PSU (the RPi is connected to a separate power source so turning off the printer does not turn off the RPi). I get the following message in Fluidd:

> Can not update MCU 'rpi' config as it is shutdown
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

This issue goes away if I just enter FIRMWARE_RESTART and I can use the printer normally afterwards. I even did a couple of prints with it and everything seems to working as normal. Do you know why I might be getting this message? It happens every time I power cycle the printer while Klipper is running (I did not have this problem prior to flashing the bootloader). It is not a big issue but it is a minor inconvenience and I would like to resolve it if possible.

I've already tried putting `restart_method: command` in the Klipper config file but to no avail.

Thanks.

GadgetAngel, 2022-01-20:
Well I use Klipper and Mainsail and anytime I shutdown the printer and leave the Raspberry Pi up and running (I use a separate Power supply for the Raspberry Pi also), I get the same message.  I have to hit the FIRMWARE_RESTART manually once the printer is back up and running. I think this question has to do more with the configuration of Fluidd and Mainsail than it does with the bootloader. You can ask on the Klipper discord channel located at https://discord.klipper3d.org

EmreDogann, 2022-01-20:
Thanks, will ask over there.

EmreDogann, 2022-01-20:
Ok so turns out I had forgotten to comment out the portion of my Klipper config that enabled ADXL Resonance Calibration. I was calibrating input shaping the day before I went ahead and flashed the bootloader.

```
# Uncomment the three sections below for ADXL345 Resonance Calibration
[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: rpi:None

[resonance_tester]
accel_chip: adxl345
probe_points:
   117.5, 117.5, 20
max_smoothing: 0.25
```

I commented out the portion of the code above in my config and now the issue is resolved.

closed 2022-01-20 by EmreDogann