GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader #8: Corrupted Bootloader - anyway to fix it?

thwerks, 2022-05-17:
I have a new Octopus 1.1 board with F446 Chip. It had firmware already loaded which i could control via octoprint but wasnt for my boards. No matter what i did it would not accept any firmware from any one of the dozen SD card i tried. I tried my own compiled firmware as well as other .bin files from other users with the same board and printer, NOTHING. never renames the file to firmware.CUR.

i read somewhere that you could use the STM32 Cube software and DFU to upload the firmware. So i tried and now i think it toast, i no longer see the device anymore when i connect in USB (no /dev/ttyACM0) no com port. I only can see the device if i put jumper on the BootD pin and reset; i tried loading the bootloader again for the F446 which succeeds but still no firmware is loaded from SD card and still no USB connection when i remove the BootD jumper and reset. 

is there anything i can do to put the boot loader back so i can connect via USB again? how can i put the original bootloader back and put a firmware on this board? maybe SD card reader isnt functioning.

GadgetAngel, 2022-05-17:
> I have a new Octopus 1.1 board with F446 Chip. It had firmware already loaded which i could control via octoprint but wasnt for my boards. No matter what i did it would not accept any firmware from [...]

I sounds like your board is totally fired, I would request a replacement board from BTT

thwerks, 2022-05-17:
ok, thank you for the input. Though is it possible to use Cube software to upload firmware rather than relying on the SD card? what would be the process for doing it that way if it were possible?

GadgetAngel, 2022-05-17:
> <GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/issues/8/1128273408@
> github.com>

I have only used the SD card to upload the firmware.  Since I have never
used the DFU method I can not advise you on that information.