GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader #10: [BUG] (SKR_MINI_E3_V3)

Putochina, 2022-07-14:
Hello 

I'm trying to upload the bootloader on a SKR MINI E3 V3 i download the file and plug the bord via ST-Link and try to upload but when i do Program and Verify it says Canot disable Read Out Protection and elf loader file is not accessible.
I can read the device memory.
I'm using the STM2 ST-Link Utility
I dont know if im doing something wrong?

GadgetAngel, 2022-07-16:
When I created the boot file from the SKR mini E3 V3 board I used the original STLINK V2 device.  Are you using an original STLINK V2 or are you using an STLINK Clone device.  

If you are using a STLINK clone my recommendation would be to buy the original STLINK V2.  The clone devices do not work sometimes and the STLINK Utility and STLINK Software use some EEPROM calls that the clone devices just do not have.

Putochina, 2022-07-16:
Hello 
I think the device is the a clone but i add a photo.
But in that way i need to buy the original one.
[image]

GadgetAngel, 2022-07-17:
> Hello I think the device is the a clone but i add a photo. But in that way i need to buy the original one. [image]

That looks like a clone STLINK.  Here are the two original STLINKS you can buy:  STLINK V2 and STLINK V3.  You can find them on Amazon.  

BUT, Before you buy the original try this one idea:

1. hold down the RESET button while connecting to the SKR mini E3 V3 to see if that allow you to connect to the board

Putochina, 2022-07-17:
Thanks for the tip but i was doing that way to conect and read the device memory i'm going to buy the original ST Link and try.

GadgetAngel, 2022-07-22:
Please review the following: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/431#issue-725132672