GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader #11: SKR 1.4T always in DFU flash mode

elhaso, 2023-01-04:
Hello,

i flashed the bootloader on my BTT 1.4 Turbo board today, with the intention to flash it via usb (hopefully via OctoPrint, but i didnt even get that far today)

straight to my problem: no matter how often i reset the board, it always starts into firmware flash mode, making it not appear in OctoPrint or Windows - the only time OctoPrint can see the board is immediately after flashing the firmware - as soon as i reset the board (powering down and restarting or simply pushing the reset button gives the same result) it goes back into flashing mode and to see it again in OctoPrint i have to DFU flash the firmware again.

I already tried:

reflashing the BTT bootloader, works as intended
reflashing the DFU bootloader (many many times)
checking for a short from PIN0.28 to GND (not the case)

Is my board fried or am i missing something here?

GadgetAngel, 2023-01-04:
Once flashing is done the SKR V1.4 Turbo should return back to operating mode.  It sounds like your board is stuck in flash mode.

Maybe you should try recompiling but use the code I branched from instead of the source code found in my repo.  I branched from https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader

Have you searched github and google to see if others have the same problem?  I found this post (maybe this can help you) at https://forum.v1engineering.com/t/skr1-4-flash-over-usb/29890/8

elhaso, 2023-01-04:
First of all, thank you for reacting that fast!

So i tried recompiling from ardiehl, no changes.
Now i measured that - once a usb cable is connected - the board always shorts 0.28 to GND - so the bootloader is not to blame since it does what it is supposed to do.

i guess there has to be something wrong with my board then. i ordered another one, so i can check for that on saturday.
still hoping to find another workaround tbh.

GadgetAngel, 2023-01-05:
> First of all, thank you for reacting that fast!
>
> So i tried recompiling from ardiehl, no changes. Now i measured that - once a usb cable is connected - the board always shorts 0.28 to GND - so the [...]

Keep me updated.

Did you try leaving the micro-SD card in the micro-SD card reader (just in case it is trying to use the sd card)?

elhaso, 2023-01-05:
Yes, i tried it with the sd card in as well as out of the board, also with a firmware.cur on it (just for the case it would behave like the original bootloader), empty SD card was tried also.

I will stay on the original bootloader for now to configure my selfbuilt printer, and will keep you updated as soon as the new board is here to try again with the DFU bootloader.

elhaso, 2023-01-18:
finally had the time to flash the bootloader to the new board - and guess what -  it works.

thanks for the help, time to close this issue! :)

closed 2023-01-18 by elhaso