GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader #13: [FR] Upload Bootloader for SKR3 and SKR3 EZ

admilsonmarques, 2023-05-25:
Hi, this week BTT sent me the hex files for the original bootload of these two boards and I'd like to add them here since this repo is well recommended for such things, do you think it is possible for me to do this?

r6e, 2024-02-02:
@admilsonmarques I could really use this; could you submit a PR here to add them? Even if they're not merged, it'd make them available for folks.

Alternatively; if you could upload them somewhere, it'd be _hugely_ appreciated. Thanks!

alakazam801, 2025-01-11:
Were these files ever made available?

GadgetAngel, 2025-10-19:
Sorry I have been ill and recovering from Brain Surgery.  If you still want to submit the files, let me know.  Just add an additional comment here.